KBNLresearch/iromlab #33: Temporary log files not written to tempDir

bitsgalore, 2017-02-24:
Applies to all wrapper modules. Solution is to change this:

    logFile = ''.join([config.tempDir,shared.randomString(12),".log"])

into:
    
    logFile = os.path.join(config.tempDir,shared.randomString(12),".log")

WillemJan, 2017-02-24:
> thread [2].
> Links:

Why not use os.tmpfile() and let the os do the moppig for you?

Anyhoe my 5 cents ;)

bitsgalore, 2017-02-24:
Good thinking, but apparently on Windows os.tmpfile() only works if the script is run as admin:

<http://stackoverflow.com/questions/7807855/os-tmpfile-doesnt-work-under-windows>

As this software will *only* be run under Windows (and most likely not as admin) I'd rather not use this.

WillemJan, 2017-02-24:
some osses I never got the gist of ;)

bitsgalore, 2017-02-24:
Fixed: <https://github.com/KBNLresearch/iromlab/commit/a0cb88c0bd4e7b399a9b34eef730de8086bd502c>

closed 2017-02-24 by bitsgalore