KBNLresearch/iromlab #66: Get rid of Carrier Type field

bitsgalore, 2018-02-09:
Reasons:

- Prone to entry errors 
- Establishing actual carrier type from physical media is not always straightforward (e.g. cd-rom vs cd-video; cd-roms that turn out to contain audio, etc.)
- Actual carrier type is already established unambiguously by cd-info

If we drop it altogether, volumeNumber sequence will simply cover all carriers that are part of a PPN (no grouping based on carrier type).

bitsgalore, 2018-08-13:
Addition/correction:

> The actual carrier type is already established automatically and unambiguously by cd-info

Not if the carrier is a DVD, which shows up as "CD-ROM with unknown filesystem" in analysis report (CD-ROMs with HFS+ FS also give this result)!!

So maybe it's better to keep the Carrier Type entry field.

robdebruin, 2019-02-05:
Even though the type detection might not be 100% proof, it would at least be useful to give the user a hint. I now find myself checking on a regular basis if an optical carrier is cd-rom or audio.

bitsgalore, 2019-02-05:
@robdebruin  I think the solution here is that the "final" establishment of the carrier type should be done further down the processing chain (i.e. outside of Iromlab). Typically this would be at the SIP creation stage. For example, the combined values  of the *containsAudio*, *containsData*, and *cdExtra* fields (derived from *cd-info*), and the *isoBuster* output (if it exists) should allow a SIP creation tool to establish the carrier type pretty much unambiguously for all cases I can think of. If so, there would be no need for the error-prone manual entry. But there might be some weird edge cases I'm overlooking for which a manually added classification would actually be helpful.

bitsgalore, 2019-05-03:
Done: <https://github.com/KBNLresearch/iromlab/commit/5e8b597016e0a9c8bb7dd580c32f8306fa156710>

closed 2019-05-03 by bitsgalore