KBNLresearch/iromlab #68: Issue with ISOBuster data extraction

timothyryanwalsh, 2018-02-22:
Hi Johan! I just sat down to try out Iromlab with our new Nimbie, having walked through the setup guide the other day (thanks for the great documentation!), and I seem to be hitting some sort of error or bug.

For any CD-ROMs I put into the Nimbie, Iromlab gets to ***Extract data session to ISO*** task, but towards the end of the ISO creation Isobuster opens an "Extracting aborted" dialog with the message "Extracting aborted. Do you want to delete the file?" (meaning the ISO file).

I was planning to dive into the code a bit to see if I can figure out why this is happening, but first I thought I'd ask if this is a familiar error?

bitsgalore, 2022-08-11:
@tw4l Hi Tessa, it's more than 4 years since you posted this, and I'm only seeing your issue now after Googling "IsoBuster" and "Extracting aborted" (for some unfathomable reason  reason Github doesn't give me notifications on issues posted in my own repos)! 

Anyway, I'm running into exactly the same issue right now with an [Iromlab spinoff](https://github.com/KBNLresearch/ipmlab/issues/15) that I want to use for imaging 3.5" floppies. Never seen this before within Iromlab with optical media, very odd. Did you ever find the cause?

bitsgalore, 2022-08-11:
Update: in Ipmlab the culprit was most likely hardware related, see <https://github.com/KBNLresearch/ipmlab/issues/15>.

tw4l, 2022-08-21:
Hi @bitsgalore, interesting, thanks for the update! I have to confess after 4 years I don't remember much about the context with this error 😅  Feel free to close the issue if you'd like!

closed 2023-11-07 by bitsgalore