KBNLresearch/iromlab #72: Commandline

robdebruin, 2019-02-04:
To be able to use Iromlab integrated into a workflow it would be neccessary to be able to call Iromlab via commandline. The following would be the minimum functionality required:

- Start Iromlab (with last batch - optional)
- Stop Iromlab
- New batch
- Finalize batch
- Submit Optical carrier (PPN, Volume number, type)
- Version of Iromlab
- Status of Iromlab (running [which PPN/volumenummer], ready, amount of errors, ...)

bitsgalore, 2019-03-28:
Don't think I completely understand this; basically this would mean completely bypassing the current UI, why would you want to do that?

Anyway, most of these functionalities (submitting a carrier, finalizing a batch, etc) are things that happen *while iromlab is already running*, and having a command line interface isn't the solution for this. I think this would require something like a [socket API](https://realpython.com/python-sockets/) where Iromlab acts a a server that listens to requests from client applications. That *might* actually be useful for allowing iromlab to communicate with e.g. an external application that manages a list of PPNs.

robdebruin, 2019-04-10:
This would make it possible to integrate into workflow. It doesn't matter how that is doen, but that it can be doen. A socket API would be fine.
We already have a workflow that keeps tracks of PPN's. We see that a lot of steps are needed to 'digitize' the disk in Iromlab and maintain the status correctly. A lot of handwork has already been eliminated, but still errors occur.

bitsgalore, 2019-05-09:
Socket interface for PPN/title entry now pretty much works: <https://github.com/KBNLresearch/iromlab/commit/182340d22aa86f7f8b90e1a73291620be2502117>

Might need some further tweaking and refactoring.

bitsgalore, 2019-05-17:
Socket API seems to work, but while making a simple demo client I ran into this:

<https://github.com/KBNLresearch/iromlab-socketclient/issues/1>

While this happens the socket API is not of much use.

bitsgalore, 2019-05-17:
Also have a look at this:

<https://docs.python.org/3/howto/sockets.html>

which looks simpler than what we have now.

bitsgalore, 2019-05-20:
Replace existing socket code by much simpler code adapted from [here](https://medium.com/python-pandemonium/python-socket-communication-e10b39225a4c), and this seems to work:

<https://github.com/KBNLresearch/iromlab/commit/b09e76a265245c0af3ae8c7800a7977929db9dd5>

closed 2019-05-27 by bitsgalore