KBNLresearch/iromlab #74: PPN Title

robdebruin, 2019-02-05:
Bij verificatie door de gebruiker wordt niet altijd de gehele titel getoond. Dit komt omdat er meerdere titel velden kunnen zijn. Bijvoorbeeld PPN 296692743.

bitsgalore, 2019-03-28:
Yep, bekend probleem. Bij meerdere titelvelden toont iromlab nu aleen de eerste. Zie [code hier](https://github.com/KBNLresearch/iromlab/blob/58bef9cc7b9d78f2cc0deccca1f6ae130c196285/iromlab/iromlab.pyw#L260). Kan wel aangepast worden (bv over alle titels binnen de lijst itereren en dan de tekstvelden aan elkaar plakken), maar dat kan ellenlange troep opleveren omdat je dan some ook weer allerlei sub-titels meekrijgt (deze velden zijn niet echt op een uniforme manier ingevuld).

De vraag lijkt me vooral of dit ook daadwerkelijk problemen oplevert bij de invoer (bv. dat het voor de gebruiker van de software niet duidelijk is of hetn om het juiste schijfje gaat)?

robdebruin, 2019-04-01:
Ja, dit kan problemen opleveren omdat men dan alleen de hoofdtitel ziet en niet de onderscheidende sub-titel. Bij schoolmateriaal komt dit vaak voor hoofdtitel: Aardrijkskunde, subtitel: groep 5/6 of subtitel: groep 7/8. Alleen op de hoofdtitel kan je geen onderscheid maken. En ja, het komt voor dat je ze allebei hebt bij het verwerken van de batch. Voor nu geen ramp omdat de workflow tool het wel toont en je zo dus zeker weet (door PPN vergelijking) dat het goed is.

bitsgalore, 2019-04-01:
OK, duidelijk, ik pas dit dan wel aan in de volgende release.

bitsgalore, 2019-04-30:
Fixed: <https://github.com/KBNLresearch/iromlab/commit/636f99cdcff377a37f09515368e18ceb4dca9b7a>

closed 2019-04-30 by bitsgalore