KBNLresearch/iromlab #75: Audio and Data (mixed mode) - Only rips audio

robdebruin, 2019-02-27:
With an optical carrier with audio and data (mixed mode) only the audio gets ripped. The data part is ignored. From batch.log:
...
2019-02-18 14:27:23,887 - INFO - ### Job identifier: 672e568c-3363-11e9-94ce-40b0341fbf5f
2019-02-18 14:27:23,887 - INFO - PPN: 136052584
2019-02-18 14:27:23,887 - INFO - Title: Tools & tales
2019-02-18 14:27:23,887 - INFO - Volume number: 1
2019-02-18 14:27:23,887 - INFO - disc directory: Z:\Work\kb-d4e78f52-335b-11e9-9c09-40b0341fbf5f\672e568c-3363-11e9-94ce-40b0341fbf5f
2019-02-18 14:27:23,902 - INFO - *** Loading disc ***
2019-02-18 14:28:11,030 - INFO - load command: C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Load\Load.exe --drive=E --rejectifnodisc --logfile=C:\temp\Iromlab\YjgmcTtZIn50.log --passerrorsback=C:\temp\Iromlab\fUv7aAD7LzuB.err
2019-02-18 14:28:11,030 - INFO - load command output: Init Nimbie: Connecting to Nimbie: Close Nimbie: Closed
2019-02-18 14:28:32,667 - INFO - *** Running cd-info ***
2019-02-18 14:28:33,026 - INFO - cd-info command: c:\users\rbr030\appdata\local\programs\python\python37\lib\site-packages\iromlab\tools\libcdio\win64\cd-info.exe -C E: --no-header --no-device-info --no-disc-mode --no-cddb --dvd
2019-02-18 14:28:33,026 - INFO - cd-info-status: 0
2019-02-18 14:28:33,026 - INFO - cdExtra: False
2019-02-18 14:28:33,026 - INFO - containsAudio: True
2019-02-18 14:28:33,042 - INFO - containsData: True
2019-02-18 14:28:33,042 - INFO - mixedMode: True
2019-02-18 14:28:33,042 - INFO - multiSession: True
2019-02-18 14:28:33,042 - INFO - *** Ripping audio ***
2019-02-18 14:30:12,492 - INFO - dBpoweramp command: C:\Program Files\dBpoweramp\kb-nl-consolerip.exe --drive=E --log=C:\temp\Iromlab\hLmxib4mhnmo.log --path=Z:\Work\kb-d4e78f52-335b-11e9-9c09-40b0341fbf5f\672e568c-3363-11e9-94ce-40b0341fbf5f
2019-02-18 14:30:12,492 - INFO - dBpoweramp-status: 0
2019-02-18 14:30:12,507 - INFO - dBpoweramp log:
Track 1: AccurateRip: Not in database   Secure: Yes   [C2B20CA8]

2019-02-18 14:30:12,507 - INFO - *** Verifying audio ***
2019-02-18 14:30:12,554 - INFO - audioHasErrors: False
2019-02-18 14:30:12,554 - INFO - Output of audio verification:
2019-02-18 14:30:12,554 - INFO - File: Z:\Work\kb-d4e78f52-335b-11e9-9c09-40b0341fbf5f\672e568c-3363-11e9-94ce-40b0341fbf5f\02.wav
2019-02-18 14:30:12,554 - INFO - File contains ID3v2 tag:    no
2019-02-18 14:30:12,554 - INFO - Data chunk block-aligned:   yes
2019-02-18 14:30:12,554 - INFO - Inconsistent header:        no
2019-02-18 14:30:12,554 - INFO - File probably truncated:    no
2019-02-18 14:30:12,554 - INFO - Junk appended to file:      no
2019-02-18 14:30:12,554 - INFO - Odd data size has pad byte: n/a
2019-02-18 14:30:12,570 - INFO - *** Writing metadata from KB-MDO to file ***
2019-02-18 14:30:12,960 - INFO - *** Computing checksums ***
2019-02-18 14:30:15,284 - INFO - *** Unloading disc ***
2019-02-18 14:30:46,422 - INFO - unload command: C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Unload\Unload.exe --drive=E --logfile=C:\temp\Iromlab\ZSVezdysseih.log --passerrorsback=C:\temp\Iromlab\zt0GCqk6MeM8.err
2019-02-18 14:30:46,422 - INFO - unload command output: Init Nimbie: Connecting to Nimbie: Close Nimbie: Closed
...

bitsgalore, 2019-03-28:
Fix would involve modifying [the code here](https://github.com/KBNLresearch/iromlab/blob/58bef9cc7b9d78f2cc0deccca1f6ae130c196285/iromlab/cdworker.py#L229) into something like this (needs testing + some optimisation on redundant bits):

```python
if carrierInfo["cdExtra"] and carrierInfo["containsData"]:
    logging.info('*** Extracting data session of cdExtra to ISO ***')
    # Create ISO file from data on 2nd session
    dirOut = dirDisc
    dataTrackLSNStart = int(carrierInfo['dataTrackLSNStart'])

    resultIsoBuster = isobuster.extractData(dirOut, 2, dataTrackLSNStart)
    statusIsoBuster = resultIsoBuster["log"].strip()
    isolyzerSuccess = resultIsoBuster['isolyzerSuccess']
    imageTruncated = resultIsoBuster['imageTruncated']

elif carrierInfo["mixedMode"] and carrierInfo["containsData"]:
    logging.info('*** Extracting data track of mixedMode disc to ISO ***')
    # Create ISO file from data on 1st session
    dirOut = dirDisc
    dataTrackLSNStart = int(carrierInfo['dataTrackLSNStart'])

    resultIsoBuster = isobuster.extractData(dirOut, 1, dataTrackLSNStart)
    statusIsoBuster = resultIsoBuster["log"].strip()
    isolyzerSuccess = resultIsoBuster['isolyzerSuccess']
    imageTruncated = resultIsoBuster['imageTruncated']
```
One thing that is odd is that cd-info reports this CD as a multisession disc (which for mixed-mode it shouldn't be), but I vaguely recall this is a bug in cd-info.

bitsgalore, 2019-05-03:
Works now: <https://github.com/KBNLresearch/iromlab/commit/0c6d7417efc32df0ca7a8817e8b88db56c0239b2>

closed 2019-05-03 by bitsgalore