KBNLresearch/iromlab #97: Add option to run Iromlab without a disc robot

bitsgalore, 2019-10-02:
This would be useful for manual procesing of carriers that failed the bulk workflow.

kieranjol, 2022-06-15:
I was wondering about this! I am only currently looking into iromlab and was hoping that it could be used without a robot, as I'm realising that the isobuster GUI doesn't do any logging at all!

bitsgalore, 2022-06-15:
@kieranjol You might want to have a look at [Iromsg](https://github.com/KBNLresearch/iromsgl), which is a single-disc / no disc robot version of Iromlab. I only created it as an interim step towards [Ipmlab](https://github.com/KBNLresearch/ipmlab), an Iromlab derivative for arbitrary portable media that I'm currently working on (this is meant to be used for imaging 3.5" floppies, and probably other storage media types later).

Since I only created this to bridge the gap between Iromlab and Ipmlab, I doubt we'll ever do any further development on Iromsgl,  but it does give you all the Iromlab functionality without needing a disc robot.

kieranjol, 2022-06-15:
Oh my god this is great! I’d no idea it existed!

kieranjol, 2022-06-15:
Apologies, I realise now that you can just run the actual scripts on their own without installation!

bitsgalore, 2022-06-15:
Yep, just clone the repo locally and then:
```  
python3 iromlab-launch.py
```
I haven't even updated setup.py so local pip install won't work. Also I'm not sure if the configuration script works, because that's just the unchanged Iromlab script. You may have to set that up manually. Or, alternatively, first install (with pip) and configure the "regular" version of Iromlab (that way you'll at least have the configuration set up in the right place), and then edit it.

kieranjol, 2022-06-15:
Your tip to create the config.xml worked. I actually don't have an optical drive right now to test with but I got as far as the 'Not a valid optical drive' warning so I think I'm on the right track! Thank you so much.

bitsgalore, 2022-06-15:
Great! For testing you could also use [WinCDEmu](https://wincdemu.sysprogs.org/) which creates a virtual optical drive.

bitsgalore, 2022-06-15:
I'll try to have a look at this tomorrow, as I haven't tested Iromsgl with the `<enablePPNLookup>False</enablePPNLookup>` setting. 

In the meantime, could you try this:

- Close Iromsgl
- In the config.xml, change the enablePPNLookup value to True (`<enablePPNLookup>True</enablePPNLookup>`)
- Re-launch the application
- In the PPN entry field, enter 155658050 (this is valid PPN identifier value)
- Hit the Submit button, and let me know what happens.

bitsgalore, 2022-06-15:
Update - this should fix it <https://github.com/KBNLresearch/iromsgl/commit/55cc25be5022951f7e46ad607e8fd324ca9e8b27>

kieranjol, 2022-06-15:
I ignored those  earlier steps and just tried out your commit and it worked! Seems like it got really far, but it dies trying to store the isobuster log:
 Terminal output
```
C:\Users\kieran.oleary\Downloads\iromsgl-main>python iromlab-launch.py
Exception in thread Thread-3 (processDisc):
Traceback (most recent call last):
  File "C:\Users\kieran.oleary\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Users\kieran.oleary\AppData\Local\Programs\Python\Python310\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\kieran.oleary\Downloads\iromsgl-main\iromlab\cdworker.py", line 214, in processDisc
    resultIsoBuster = isobuster.extractData(dirDisc, 1, 0)
  File "C:\Users\kieran.oleary\Downloads\iromsgl-main\iromlab\isobuster.py", line 46, in extractData
    with io.open(logFile, "r", encoding="cp1252") as fLog:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\kb-87e5f0d6-ecd9-11ec-8263-909c4aba942c\\8d7a7f65-ecd9-11ec-a19b-909c4aba942c\\isobuster.log'
```

Config file

```
<?xml version="1.0"?>
<!-- iromlab configuration file. This file MUST be in the same directory 
as iromlab.py/ iromlab.exe! 
-->

<config>

<!-- Note on file paths: do NOT wrap any of the file paths below in quotes, even if they contain
   spaces! (the Python os.path libs don't seem to like this!)
-->

<!-- CD drive letter -->
<cdDriveLetter>E</cdDriveLetter>

<!-- root directory - this is the default search path for creating / opening batches -->
<rootDir>C:\</rootDir>

<!-- directory for storing temporary files -->
<tempDir>C:\Temp</tempDir>

<!-- prefix that is used to create batch names -->
<prefixBatch>kb</prefixBatch>

<!-- String that sets audio format. Permitted values: wav or flac. ONLY used for the audio verification, 
the ripping format must be set from dBpoweramp's CD Ripper tool (GUI) defined in the Windows registry  -->
<audioFormat>flac</audioFormat>

<!-- maximum number of seconds that iromlab will wait while trying to load a new disc 
this will prevent iromlab from entering an infinite loop if e.g. a disc cannot be loaded
properly because its is badly damaged
-->
<secondsToTimeout>20</secondsToTimeout>

<!-- host address that is used if socket API is enabled. Use 127.0.0.1 for localhost
-->
<socketHost>127.0.0.1</socketHost>

<!-- port that is used if socket API is enabled.
-->
<socketPort>65432</socketPort>

<!-- flag that -if False- disables PPN lookup. If False, PPN is set to empty string and
operator must enter descriptive Title field in UI
-->
<enablePPNLookup>False</enablePPNLookup>

<!-- flag that -if True- causes Iromlab to wait processing any discs until the operator
finalises the batch 
-->
<startOnFinalize>False</startOnFinalize>
[... 9 lines cut ...]
-->
<reportFormatString><![CDATA[{'DFXML (IsoBuster 4.2 version)'}{%UTF8}{%DFXML}{%GMT}{%FOLDERS}{%STREAMS}<%XMLHEADER><%BR><dfxml xmlns='http://www.forensicswiki.org/wiki/Category:Digital_Forensics_XML'<%BR> xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'<%BR> xmlns:dc='http://purl.org/dc/elements/1.1/'<%BR> xmlns:hfs='http://www.forensicswiki.org/wiki/HFS' version='1.0'><%BR><%BR> <metadata><%BR> <dc:type><%DEVICETYPE></dc:type><%BR> </metadata><%BR><%BR> <creator><%BR> <program><%APP></program><%BR> <version><%VERSION></version><%BR> <execution_environment><%BR> <start_time><%SYSTIMEDATE></start_time><!--GMT--><%BR> <os_version><%OS></os_version><%BR> <username><%USER></username><%BR> </execution_environment><%BR> </creator><%BR><%BR> <source><%BR> <device_model><%DEVICE></device_model><%BR> <image_filename><%DEVICEPATH></image_filename><%BR> <image_size><%DEVICEFILESIZE></image_size><%BR> <sectorsize><%DEVICEBLOCKSIZE></sectorsize><%BR> <devicesectors coding='base10'><%DEVICEBLOCKS></devicesectors><%BR> </source><%BR><%BR> <volume><%BR> <partition_offset><%PARTITIONLBABYTESOFFSET></partition_offset><%BR> <ftype_str><%TYPE></ftype_str>{%HEADER}{%FOLDER}<%BR> <fileobject><%BR> <filename><%RELPATH></filename><%BR> <name_type>d</name_type><%BR> <filesize><%BYTES></filesize><%BR> <alloc>1</alloc><%BR> <inode><%UID></inode><%BR> <mtime><%TIMEDATE></mtime><!--GMT--><%BR> <byte_runs><%EXTENTLOOP> </byte_runs><%BR> </fileobject>{%FILE}<%BR> <fileobject><%BR> <filename><%RELPATH></filename><%BR> <name_type>r</name_type><%BR> <filesize><%BYTES></filesize><%BR> <alloc>1</alloc><%BR> <inode><%UID></inode><%BR> <mtime><%TIMEDATE></mtime><!--GMT--><%BR> <hfs:HFStype_creator><%TYPE>/<%CREATOR></hfs:HFStype_creator><!--Only relevant if MAC File System--><%BR> <byte_runs><%EXTENTLOOP> </byte_runs><%BR> </fileobject>{%STREAM}<%BR> <fileobject><!--Stream or Resource Fork--><%BR> <filename><%RELPATH></filename><%BR> <name_type>-</name_type><%BR> <filesize><%BYTES></filesize><%BR> <alloc>1</alloc><%BR> <inode><%UID></inode><%BR> <mtime><%TIMEDATE></mtime><!--GMT--><%BR> <byte_runs><%EXTENTLOOP> </byte_runs><%BR> </fileobject>{%EXTENT} <byte_run img_offset='<%LBABYTEOFFSET>' len='<%BYTES>' />{%FOOTER} </volume><%BR><%BR> <runstats><%BR> <stop_time><%SYSTIMEDATE></stop_time><!--GMT--><%BR> <clock_seconds><%SYSTIMELAPSEDSEC></clock_seconds><%BR> </runstats><%BR><%BR></dfxml><%BR><!-- For more information: https://www.isobuster.com/reports -->]]></reportFormatString>

<!-- Below items point to the locations of all executables that are wrapped by Iromlab
-->

<!-- location of Nimbie drivers -->
[... cut ...]
<isoBusterExe>C:\Program Files (x86)\Smart Projects\IsoBuster\IsoBuster.exe</isoBusterExe>

<!-- location of dBpoweramp console ripper -->
<dBpowerampConsoleRipExe>C:\Program Files\dBpoweramp\kb-nl-consolerip.exe</dBpowerampConsoleRipExe>

</config>
```
Folder created sans ISO:

[kb-87e5f0d6-ecd9-11ec-8263-909c4aba942c.zip](https://github.com/KBNLresearch/iromlab/files/8912168/kb-87e5f0d6-ecd9-11ec-8263-909c4aba942c.zip)

and the iso I used for testing was: http://archive.ubuntu.com/ubuntu/dists/bionic/main/installer-amd64/current/images/netboot/mini.iso

kieranjol, 2022-06-15:
Actually it's weird, that file is indeed there and it has just a value of 0. No idea why python can't open it.
Tried this seperately

```
>>> a =  'C:\\kb-87e5f0d6-ecd9-11ec-8263-909c4aba942c\\8d7a7f65-ecd9-11ec-a19b-909c4aba942c\\isobuster.log'
>>> import os
>>> os.path.isfile(a)
True
>>> import io
>>> with io.open(a, "r", encoding="cp1252") as fLog:
...     log = fLog.read()
...
>>> print(log)
0
```

bitsgalore, 2022-06-15:
I ran into that with Ipmlab as well (fixed that), so I just applied the same fix to Iromsgl: <https://github.com/KBNLresearch/iromsgl/commit/2e61bd252d9a67d1ab49f2eebcaacbc1a1b4d592>

Let me know if this works! (Very strange error though, and I don't quite understand how this is even possible. Perhaps the log file is created by some subprocess of Isobuster that hasn't quite finished yet by the time my code tries to open it, no idea!)

kieranjol, 2022-06-15:
Your commit worked! I needed to add one more config call and then I think I finally got to the end of the process!
https://github.com/KBNLresearch/iromsgl/pulls#4

bitsgalore, 2022-06-16:
Perfect, I just merged your pull request (and applied [the same fix to Ipmlab](https://github.com/KBNLresearch/ipmlab/commit/3aa5398e1faab57df3da1e64ea141fa0a0f96fc5), which has the same problem).

bitsgalore, 2022-06-16:
By the way, while working on the Ipmlab code I made several changes/improvements to the UI behaviour (especially to avoid strange things happening when you press the Quit button while some process is still running, etc.). I will probably patch those back into Iromsgl at some point, but that might take a while (I don't think I've quite nailed it yet in Ipmlab).

kieranjol, 2022-06-16:
It would be great to get those patched back. Is there any chance that lpmlab could get the non robot functionality of iromsgl? Or perhaps there iromlab gets that functionality? I’d make a stab at doing so for iromlab if I thought it was doable.

bitsgalore, 2022-06-16:
Ipmlab is non-robot by default, because I'm not aware of any robot devices for floppy disks (or other portable media). So basically Ipmlab is really similar to Iromsgl, but just without the optical-media specific stuff.