KBNLresearch/iromlab #100: ImportError: No system module 'pywintypes' (pywintypes38.dll)

bitsgalore, 2020-03-04:
Install of Iromlab on new Windows 10 machine results in following error when running iromlab-launch.py of iromlab-configure.py from source repo:

```
ImportError: No system module 'pywintypes' (pywintypes38.dll)
``` 

For a pip install the config and launch script simply run without anything happening.

bitsgalore, 2020-03-04:
See also:

<https://stackoverflow.com/questions/18907889/importerror-no-module-named-pywintypes>
and

<https://stackoverflow.com/questions/25254285/pyinstaller-importerror-no-system-module-pywintypes-pywintypes27-dll>.

Made a check for package pypiwin32, which is is actually installed. Following the recommendation [here](https://stackoverflow.com/a/34123049) I looked at this folder:

```
C:\Users\JKN010\AppData\Roaming\Python\Python38\site-packages\pywin32_system32
```

I then copied the files 'pythoncom38.dll' and 'pywintypes38.dll' over to:

```
C:\Users\JKN010\AppData\Roaming\Python\Python38\site-packages\win32\lib 
```

After this everything works OK. Ugly hack though, there must be a cleaner way to do this?

Side note: worth mentioning that so far I've only tried doing a user install of Iromlab and its dependencies. Should test what happens with a global install.

minecraft3r217, 2020-05-28:
very odd had the same problem will try your solution

JohnS-coder, 2020-07-04:
Interesting but it really works.. Thank you very much.. Saved my hours.

meicheerze, 2020-09-10:
Thank you so much @bitsgalore for me this error occured in pyttsx3 library and this same hack worked

rajsingh010, 2020-10-16:
> win32\lib

thanks it resolves my issue

Mayur710, 2020-12-01:
Thank you 
Problem solved :)

yogeshk04, 2020-12-08:
@bitsgalore Thank you

Redshift1949, 2020-12-18:
Thank you this worked for me

extorc, 2021-01-02:
OHMG worked

Red-Eyed, 2021-01-11:
@bitsgalore Thanks, your workaround pointed me out the reason of the problem.  
Alternative workaround w/o copying, is just setting `PATH` at the beginning:    
```python
from pathlib import Path
import os
import sys
prefix = Path(sys.exec_prefix).resolve()
os.environ["PATH"] = f"{prefix / 'Lib\\site-packages\\pywin32_system32'}{os.pathsep}{os.environ['PATH']}"
```

TrinhtrungTH, 2021-02-15:
it worked
 thanks so much!

KADIAPRIYANSHU, 2021-06-16:
thank you so much!

jorgelex008, 2021-06-27:
Excellent! Also worked for me like a charm. Many thanks!! :-)
(Note: Issue related to pyttsx3).

mohamad22mk, 2021-07-19:
Thank you very much very much

bitsgalore, 2023-12-04:
Update - [this update](https://stackoverflow.com/a/73576949) to the aforementioned stackoverflow thread suggests things may have changed in Windows 11:

> I typically import pywintypes or use a library that does. This import suddenly stopped working following update to Windows 11.
> Using PIP I uninstalled pywin32, then reinstalled it (into my global python as above). This worked, I can do the import now. Just saying cause this is a simple step

Needs further testing/confirmation (don't have access to a Win 11 environment at the moment).

mehclere, 2024-04-16:
> C:\Users\JKN010\AppData\Roaming\Python\Python38\site-packages\win32\lib 

I'm trying to do a user install of IROMLAB and am experiencing this issue, despite moving these files. Running Windows 10, python 3.12.3 (files were named pythoncom312.dll and pywintypes312.dll, not sure if that matters). Please let me know if there's any information I can provide to help troubleshoot this. I'm really bummed I can't get IROMLAB up and running!

mehclere, 2024-04-16:
> Made a check for package pypiwin32, which is is actually installed. Following the recommendation [here](https://stackoverflow.com/a/34123049) I looked at this folder:
>
> ``` [...]

I'm trying to do a user install of IROMLAB and am experiencing this issue, despite moving these files. Running Windows 10, python 3.12.3 (files were named pythoncom312.dll and pywintypes312.dll, not sure if that matters). Please let me know if there's any information I can provide to help troubleshoot this. I'm really bummed I can't get IROMLAB up and running!

bitsgalore, 2024-04-16:
@mehclere  Not 100% sure, but this is probably related to this issue (and if it's not, you'll eventually run into it anyway):

<https://github.com/KBNLresearch/iromlab/issues/107>

As a workaround, try if you can make it work with Python 3.11 or earlier (you may need to apply the dll fixes).

Meanwhile I'll have a look at how to remove the imp dependency, hopefully I'll get round to this one of these weeks.

Other thing to watch out for, is that I've heard some reports of issues when using Iromlab in conjunction with the latest (5.x) versions of IsoBuster. To be on the safe side I'd recommend to use the most recent 4.x release of IsoBuster (4.9, if I recall correctly). This is also something I'll try to sort out at some point, but this probably won't happen anytime soon.

mehclere, 2024-04-17:
@bitsgalore thank you so much for your quick response, using the earlier version of Python worked! I will inform you if I run into any issues with IsoBuster 5.3.

bitsgalore, 2024-04-17:
@mehclere See also <https://github.com/KBNLresearch/iromlab/issues/109>; in particular if you run into any unexplained IsoBuster 1005 errors you might want to give IsoBuster 4.9 a try. IsoBuster's author did some major refactoring of the codebase in version 5.0 so there could be some more surprises (especially because I don't think many IsoBuster users use its command-line interface, which is how Iromlab wraps it internally).

mford99, 2024-06-10:
Thank you! Saved me hours