KBNLresearch/iromlab #105: Cannot install Iromlab via PIP

Hf23d4, 2023-05-03:
Hey. Just got a Nimbie and have gone through all of the Isobuster and DBPowerAmp install elements but am stuck at the section that deals with installing Iromlab via PIP. I've tried putting various file paths and nothing seems to be working. Suggestions on how to resolve this?

kieranjol, 2023-05-04:
Hi! Can you post the error that you receive?

Hf23d4, 2023-05-04:
It varies, but in general it says that it is not the correct command. I've no doubt that I'm not putting in the command correctly due to being unsure of where the correct Python file is (there are many that show up in the C drive).

kieranjol, 2023-05-04:
What is the output of “pip install iromlab”
Can you paste the output here or a screenshot?

Hf23d4, 2023-05-04:
Hey. Was actually working on it the past few minutes and seem to have installed it. This is what showed up on the screen:

C:\Users\Keith>C:\Users\Keith\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\pip install --user iromlab
Collecting iromlab
  Downloading iromlab-1.0.8-py2.py3-none-any.whl (2.8 MB)
     |████████████████████████████████| 2.8 MB 2.2 MB/s
Collecting requests
  Downloading requests-2.29.0-py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 4.5 MB/s
Collecting pypiwin32
  Downloading pypiwin32-223-py3-none-any.whl (1.7 kB)
Requirement already satisfied: setuptools in c:\program files\windowsapps\pythonsoftwarefoundation.python.3.7_3.7.2544.0_x64__qbz5n2kfra8p0\lib\site-packages (from iromlab) (47.1.0)
Collecting lxml
  Downloading lxml-4.9.2-cp37-cp37m-win_amd64.whl (3.8 MB)
     |████████████████████████████████| 3.8 MB 6.4 MB/s
Collecting isolyzer>=1.3.1
  Downloading isolyzer-1.4.0-py2.py3-none-any.whl (36 kB)
Collecting wmi
  Downloading WMI-1.5.1-py2.py3-none-any.whl (28 kB)
Collecting charset-normalizer<4,>=2
  Downloading charset_normalizer-3.1.0-cp37-cp37m-win_amd64.whl (94 kB)
     |████████████████████████████████| 94 kB 1.2 MB/s
Collecting idna<4,>=2.5
  Downloading idna-3.4-py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB ...
Collecting certifi>=2017.4.17
  Downloading certifi-2022.12.7-py3-none-any.whl (155 kB)
     |████████████████████████████████| 155 kB 6.4 MB/s
Collecting urllib3<1.27,>=1.21.1
  Downloading urllib3-1.26.15-py2.py3-none-any.whl (140 kB)
     |████████████████████████████████| 140 kB 6.4 MB/s
Collecting pywin32>=223
  Downloading pywin32-306-cp37-cp37m-win_amd64.whl (9.3 MB)
     |████████████████████████████████| 9.3 MB 3.3 MB/s
Installing collected packages: charset-normalizer, idna, certifi, urllib3, requests, pywin32, pypiwin32, lxml, isolyzer, wmi, iromlab
  WARNING: The script normalizer.exe is installed in 'C:\Users\Keith\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script isolyzer.exe is installed in 'C:\Users\Keith\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed certifi-2022.12.7 charset-normalizer-3.1.0 idna-3.4 iromlab-1.0.8 isolyzer-1.4.0 lxml-4.9.2 pypiwin32-223 pywin32-306 requests-2.29.0 urllib3-1.26.15 wmi-1.5.1
WARNING: You are using pip version 20.1.1; however, version 23.1.2 is available.
You should consider upgrading via the 'C:\Users\Keith\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\python.exe -m pip install --upgrade pip' command.

The next step should be to go to the iromlab-configure.exe file but when that is clicked, nothing happens in the command prompt.

Hf23d4, 2023-05-04:
It just goes to the command prompt and does nothing. Shows a blinking cursor next to the C:\Users\Keith prompt.

kieranjol, 2023-05-04:
Looking at the docs, it says if the config fails and you don’t get the pop
up saying all went well, you can try this workaround?
https://github.com/KBNLresearch/iromlab/issues/100#issuecomment-594656069

kieranjol, 2023-05-04:
Btw did you install python from the windows App Store? Your paths look like that might be the case. I prefer uninstalling windows App Store python and instead  installing python from the python website and making sure that I tick the box at the very start to add python to PATH. It tends to work with pip so much better than the windows App Store version which doesn’t add pip to path. This might not solve your configure issue but it might lead to an easier time with python. https://www.python.org/downloads/

Hf23d4, 2023-05-04:
It's possible I'm hitting the wrong .exe file, though, as that workaround is a for a specific error. When I hit the iromlab-configure.exe file (or at least what I think is that file), it just pulls up a command prompt and nothing happens.

Hf23d4, 2023-05-04:
I used the Python software that came with the computer. This is so far out of my wheelhouse, may just have to try to find another way to get the Nimbie working or send it back (it should come with better software than it does).

kieranjol, 2023-05-04:
Johan (the developer) might be able to provide  assistance, I’m just a casual user and my troubleshooting skills are limited!

Hf23d4, 2023-05-04:
I really appreciate all of your time and help. Maybe looking at it with a fresh set of eyes on this end will help :-)

bitsgalore, 2023-05-04:
@Hf23d4 Can you post a screenshot that shows what happens if you double-click iromlab-configure.exe? Does it start a window like the one below (note the "Iromlab Configuration Tool" text):

[image]

Or do you see something else? Also, what Windows version are you using?

On an unrelated note: I've noticed that with the latest (5.x) IsoBuster versions, Iromlab sometimes throws an internal FileNotFoundError on the IsoBuster log file (which then causes Iromlab to become unresponsive). Since we're still using an older IsoBuster version internally this isn't a problem for us, but it will most likely be for you. Because of this I just added a patch that fixes this. To  apply the patch, try re-installing Iromlab like this:

```
C:\Users\Keith>C:\Users\Keith\AppData\Local\Microsoft\WindowsAppsPythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\pip install --upgrade --user iromlab
```
Then pay close attention to the console output, which should show a reference to the 1.0.9 version of Iromlab:

> Downloading iromlab-1.0.8-py2.py3-none-any.whl (2.8 MB)

If you still see the old (1.08) version, just re-run the command.

Hf23d4, 2023-05-04:
Hey, @bitsgalore. Thanks for the info. I can't seem to get the command line instructions to work properly and the Nimbie is ending up being too much of a programming task to move ahead. Have packed up the Nimbie and am returning it and going with more of a plug-and-play manual DVD transfer route.

Thanks again for the assistance.

bitsgalore, 2023-05-08:
@Hf23d4 Thanks for the update, I'll close the issue then.

closed 2023-05-08 by bitsgalore