KBNLresearch/iromlab #109: Nimbie loading/rejecting, #1005 error

darcyruppert, 2024-03-25:
Hello,

I've been testing an IROMLAB/Nimbie workflow on a collection of CD-Rs at the museum I work at. I have successfully created disc images, but each batch manifest reports success as FALSE, with a 1005 error in the Isobuster log. Each disc also takes a long time to process, seemingly because of long pauses before loading and rejecting discs. I've attached an example batch log, where you can see that 10 minutes pass between the "*** Loading disc ***" notification and the actual load command:

> 2024-03-25 11:01:51,852 - INFO - *** Loading disc ***
> 2024-03-25 11:11:38,832 - INFO - load command: C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Load\Load.exe --drive=F --rejectifnodisc --logfile=C:\Temp\6NhBdJv1RuQ8.log --passerrorsback=C:\Temp\GEaKXfQnfws5.err

Rejecting the disc is similar, with 5 minutes passing between notification and command. In total it takes 20 minutes to process 1 disc. Creating disc images in Isobuster by itself takes roughly 3 minutes. I'm not sure if the "success=FALSE" and 1005 error are related to the long processing time, but I don't have any other leads. 

My config:
<?xml version="1.0"?>

<config>

<cdDriveLetter>F</cdDriveLetter>

<rootDir>C:\Digital Preservation\test032524</rootDir>

<tempDir>C:\Temp</tempDir>

<prefixBatch>mohai</prefixBatch>

<audioFormat>wav</audioFormat>

<secondsToTimeout>20</secondsToTimeout>

<socketHost>127.0.0.1</socketHost>

<socketPort>65432</socketPort>

<enablePPNLookup>False</enablePPNLookup>

<startOnFinalize>True</startOnFinalize>

<enableSocketAPI>False</enableSocketAPI>

<reportFormatString><![CDATA[{'DFXML (IsoBuster 4.2 version)'}{%UTF8}{%DFXML}{%GMT}{%FOLDERS}{%STREAMS}<%XMLHEADER><%BR><dfxml xmlns='http://www.forensicswiki.org/wiki/Category:Digital_Forensics_XML'<%BR> xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'<%BR> xmlns:dc='http://purl.org/dc/elements/1.1/'<%BR> xmlns:hfs='http://www.forensicswiki.org/wiki/HFS' version='1.0'><%BR><%BR> <metadata><%BR> <dc:type><%DEVICETYPE></dc:type><%BR> </metadata><%BR><%BR> <creator><%BR> <program><%APP></program><%BR> <version><%VERSION></version><%BR> <execution_environment><%BR> <start_time><%SYSTIMEDATE></start_time><%BR> <os_version><%OS></os_version><%BR> <username><%USER></username><%BR> </execution_environment><%BR> </creator><%BR><%BR> <source><%BR> <device_model><%DEVICE></device_model><%BR> <image_filename><%DEVICEPATH></image_filename><%BR> <image_size><%DEVICEFILESIZE></image_size><%BR> <sectorsize><%DEVICEBLOCKSIZE></sectorsize><%BR> <devicesectors coding='base10'><%DEVICEBLOCKS></devicesectors><%BR> </source><%BR><%BR> <volume><%BR> <partition_offset><%PARTITIONLBABYTESOFFSET></partition_offset><%BR> <ftype_str><%TYPE></ftype_str>{%HEADER}{%FOLDER}<%BR> <fileobject><%BR> <filename><%RELPATH></filename><%BR> <name_type>d</name_type><%BR> <filesize><%BYTES></filesize><%BR> <alloc>1</alloc><%BR> <inode><%UID></inode><%BR> <mtime><%TIMEDATE></mtime><%BR> <byte_runs><%EXTENTLOOP> </byte_runs><%BR> </fileobject>{%FILE}<%BR> <fileobject><%BR> <filename><%RELPATH></filename><%BR> <name_type>r</name_type><%BR> <filesize><%BYTES></filesize><%BR> <alloc>1</alloc><%BR> <inode><%UID></inode><%BR> <mtime><%TIMEDATE></mtime><%BR> <hfs:HFStype_creator><%TYPE>/<%CREATOR></hfs:HFStype_creator><%BR> <byte_runs><%EXTENTLOOP> </byte_runs><%BR> </fileobject>{%STREAM}<%BR> <fileobject><%BR> <filename><%RELPATH></filename><%BR> <name_type>-</name_type><%BR> <filesize><%BYTES></filesize><%BR> <alloc>1</alloc><%BR> <inode><%UID></inode><%BR> <mtime><%TIMEDATE></mtime><%BR> <byte_runs><%EXTENTLOOP> </byte_runs><%BR> </fileobject>{%EXTENT} <byte_run img_offset='<%LBABYTEOFFSET>' len='<%BYTES>' />{%FOOTER} </volume><%BR><%BR> <runstats><%BR> <stop_time><%SYSTIMEDATE></stop_time><%BR> <clock_seconds><%SYSTIMELAPSEDSEC></clock_seconds><%BR> </runstats><%BR><%BR></dfxml><%BR>]]></reportFormatString>

<prebatchExe>C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Pre-Batch\Pre-Batch.exe</prebatchExe>
<loadExe>C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Load\Load.exe</loadExe>
<unloadExe>C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Unload\Unload.exe</unloadExe>
<rejectExe>C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Reject\Reject.exe</rejectExe>

<isoBusterExe>C:\Program Files\Smart Projects\IsoBuster\IsoBuster.exe</isoBusterExe>

<dBpowerampConsoleRipExe>C:\Program Files\dBpoweramp\kb-nl-consolerip.exe</dBpowerampConsoleRipExe>

</config>

[batch.log](https://github.com/KBNLresearch/iromlab/files/14748556/batch.log)

bitsgalore, 2024-03-25:
What IsoBuster version are you using? I came across some unexplained 1005 errors (which usually indicate the user manually interrupted the imaging, which obviously isn't the case here) while working with IsoBuster 5.x on a spinoff project for floppy discs:

<https://github.com/KBNLresearch/ipmlab/issues/15>

Since I ended up using something else there for the floppy imaging I never followed this up. For our operational workflow we're still using IsoBuster 4 (don't remember the exact version), and I haven't tested it with version 5. This might explain part of the issue.

But it doesn't explain the 5 minute delay after the reject command is issued: 

```
2024-03-25 11:14:48,518 - INFO - *** Rejecting disc ***
2024-03-25 11:19:47,143 - INFO - reject command: C:\Program Files\dBpoweramp\BatchRipper\Loaders\Nimbie\Reject\Reject.exe --drive=F --logfile=C:\Temp\WTDADT3dFhdq.log --passerrorsback=C:\Temp\LyWgDsJyTDTz.err
```

This makes me wonder if this might be some hardware issue (most likely related to the Nimbie's optical drive). 

Also, I remember we had some similar problems (at least unexplained delays) years ago after we had first set up our operational workflow. From what I recall, those were related to the drive configuration in dbPoweramp (explained [here](https://github.com/KBNLresearch/iromlab/blob/master/doc/setupDbpoweramp.md#drive-configuration)). I can't find exactly how we fixed that, but from what I remember we managed to get things working by re-configuring the drive.

Let me know if this helps!

bitsgalore, 2024-03-25:
Come to think of it, I *think* the problem we had at the time was caused by the presence of multiple configured drives, which somehow caused the load/unload/reject driver tools to get "stuck" on the wrong drive. So make sure that only one drive is configured, and also make sure "Prevent auto-run on all CD drives" is checked.

bitsgalore, 2024-03-26:
Other thing that crossed my mind - I think I actually solved our slow response issue of the Nimbie drivers at the time by disabling any unused optical drives (including the PC's internal one) from the Windows device manager. See screenshot in below post:

<https://forums.tomshardware.com/threads/how-to-stop-cdrom-drive-from-work-after-sleep-mode.3630050/post-21877311>

Also note that after disabling one device, Windows may re-map existing devices (like the Nimbie's optical drive) to another logical drive letter (e.g. from "F" to "E" or something).

So I'd suggest that after disabling any devices, you:

1. reboot your machine (with the Nimbie switched on)
2. check for any changes to the logical drive mappings,
3. update your configuration (Iromlab config file + dBPoweramp config) accordingly, if needed.

darcyruppert, 2024-03-26:
There was a second drive configured -- after unconfiguring I was able to process three discs in 12 minutes! Thanks for your help. 

I am still running into the 1005 error, so I may try uninstalling Isobuster (I'm running 5.3, the most recent version) and installing 4.0.

bitsgalore, 2024-03-26:
@darcyruppert Good to hear the Nimbie drivers are now working properly. As for Isobuster: I would try running the most recent 4.x version (I think this was 4.7, not entirely sure what we're using for our operational workflow but I can ask). You mention 4.0, but that version is probably too old (Iromlab uses a couple of features that were added later). 

I know IsoBuster's author did some pretty major refactoring of the code for the 5.0 release, so it could be that this introduced some unwanted effects. At some point I'd like to upgrade Iromlab to IB 5,  and by that time I'll give this a more in-depth look. But this won't happen anytime soon.

darcyruppert, 2024-03-26:
@bitsgalore Ah yes, I was just noticing that myself and thinking of downloading 4.9: https://www.isobuster.com/news/isobuster_4.9_release_notes 

But let me know if there's a version that's most compatible with Iromlab. I've also been having some issues using Isobuster 5.3 on old Mac discs with HFS file systems, so I've been thinking of experimenting with older versions.

bitsgalore, 2024-03-26:
@darcyruppert 4.9 should be fine. Let me know if you keep getting this issue and I'll try to think of something!

darcyruppert, 2024-03-26:
Will do, thanks so much!

bitsgalore, 2024-03-27:
For future reference, I just added a section on disabling unused optical drives to the Iromlab setup documentation:

<https://github.com/KBNLresearch/iromlab/blob/master/doc/setupIromlab.md#disable-unused-optical-drives>

And also a reference to this in the User Guide's Troubleshooting section:

<https://github.com/KBNLresearch/iromlab/blob/master/doc/userGuide.md#loading-unloading-or-rejecting-a-disc-takes-a-very-long-time>

CyberGonzo, 2024-04-24:
> I know IsoBuster's author did some pretty major refactoring of the code for the 5.0 release, so it could be that this introduced some unwanted effects. At some point I'd like to upgrade Iromlab to IB 5, and by that time I'll give this a more in-depth look. But this won't happen anytime soon.

If there's anything I can do in the short term, do let me know.
IsoBuster 5.4 is due ( https://www.isobuster.com/news/isobuster_5.4_beta_release_notes ) in a little while.  If I can fix or improve something before IsoBuster 5.4 (final) is released, now is the moment ;)

bitsgalore, 2024-04-24:
@CyberGonzo Thanks for the heads-up and the offer! I just sent you a quick email with an outline of our plans with Iromlab, and how IsoBuster fits into this.

closed 2024-09-05 by bitsgalore