KBNLresearch/iromlab #115: kb-nl-consolerip secure settings

Jaubr, 2024-07-31:
Hello. Does kb-nl-consolerip support any secure settings that are in the GUI version, such as "Drive Speed", "Minimum Ultra Passes", "Maximum Ultra Passes", "End After Clean Passes" and others that are in the GUI. If it does, please give me a link or write them to me. I assume that they can be written in the copy command, for example:

kb-nl-consolerip --drive="D" --log="\log.txt" --path="" **--drive_speed="40" --minimum_ultra_passes="2" ...**

I ask this question because copying audio CDs via console is 2-3 times faster for me than via GUI, this means that perhaps when copying in the console, the above settings do not work (although they are specified in the GUI version at this time).
Thanks.

bitsgalore, 2024-07-31:
Assuming that the specific settings you mention can be set in either the Batch ripper configuration or CD Ripper GUI applications, you can just use those. Running kb-nl-consolerip will then use those settings. See also the instructions on configuring dBPoweramp here:

<https://github.com/KBNLresearch/iromlab/blob/master/doc/setupDbpoweramp.md>

Jaubr, 2024-08-01:
So, in theory, kb-nl-consolerip should automatically use the settings specified in CD Ripper GUI?

bitsgalore, 2024-08-01:
@Jaubr Yes!

Jaubr, 2024-08-02:
Then why does copying through the GUI take, on average, say, 10 minutes, but through kb-nl-consolerip with the same settings it often takes 3 minutes?

Jaubr, 2024-08-02:
I think copying can't take 3 minutes if the settings say to do 2 to 4 ultra passes. Through the GUI I see that it does 2-4, so it takes about 10 minutes, and through kb-nl-consolerip judging by the speed of the process it seems to me that it only does 1 pass.