KBNLresearch/iromlab #122: Replace IsoBuster and DbPoweramp with redump

bitsgalore, 2024-11-10:
[Redump.org](http://www.wiki.redump.org/index.php?title=Disc_Dumping_Guide_(MPF)) has updated its earlier IsoBuster recommendation ([here](http://www.wiki.redump.org/index.php?title=DVD-Video_Dumping_Guide)) to [MPF](https://github.com/SabreTools/MPF), which is a front-end for Aaru and [redumper](https://github.com/superg/redumper).

Redumper gets a recommendation in [this guide by Misty De Meo](https://www.mistys-internet.website/blog/blog/2024/09/13/the-working-archivists-guide-to-enthusiast-cd-rom-archiving-tools/#fnref:1). It is able to handle both audio and data discs.

Combined with <https://github.com/KBNLresearch/iromlab/issues/110> this could potentially remove all proprietary software, and also make Iromlab cross-platform.

kieranjol, 2024-11-10:
This would be amazing!

bitsgalore, 2024-11-10:
@kieranjol  No promises yet  this will actually happen though, mainly created this ticket as a reminder to myself. I also have zero experience with Redumper, but it looks like a really interesting option. And those proprietary/Windows-only restrictions have always been a bit of a bugbear to me.

bitsgalore, 2024-12-18:
In addition, I just came across [Whipper](https://github.com/whipper-team/whipper), which is "a Python 3 (3.6+) CD-DA ripper based on the [morituri project](https://github.com/thomasvs/morituri) (CDDA ripper for *nix systems aiming for accuracy over speed)". This could potentially be a replacement for dBPoweramp.